suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config #1: Add TPU filament

kyelisavenko, 2023-10-06:
First and foremost, I much appreciate your profiles. You are the timesaver for a lot of people. Thanks.
I want to ask if there is a chance to see a TPU filament configuration in the nearest perspective. It would be great to have it in addition to the most popular PLA and PETG.
Thanks in advance.

suchmememanyskill, 2023-10-06:
I can try to look into it, sure. I ordered a roll of TPU so i can try to make a profile around it. Do note i am unavailable for the next few weeks so it may take a while before i get around to this. 

Although i suspect a good starting point would be to start from a PLA profile and limit the volumetric flow rate to 2 or 3mm^3/s and decreasing the retraction speed to around 30mm/s

kyelisavenko, 2023-10-06:
You don't need to rush at all. Will try some calibration details based on your recommendations. Thanks

suchmememanyskill, 2023-10-06:
If you can find a reasonable setup, would you mind sending a pull request with the profile?

suchmememanyskill, 2023-11-15:
Can someone else try my TPU profile? You can either copy the .ini manually or use the auto generated config bundle: https://github.com/suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config/actions/runs/6871042960

closed 2023-12-09 by suchmememanyskill