suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config #9: Maker's Muse slicer setting optimisations

MarkusNemesis, 2023-11-11:
Firstly, thank you so much for making this profile. I've been using it the past three weeks and loving it as a first time 3d printer user.

Secondly, now that you've got a heap of presets for various layer heights, I find myself having to apply a bunch of changes to them all each time this repo is updated, and was wondering if you'd want to integrate these 'optimisations' to your profile?

Here's the Maker's Muse video I reference:
https://www.youtube.com/watch?v=mE521Q4H6aY

tl;dw of the setting recommendations:
Perimeters: 3 (from 2)
Infil: Cubic (from Grid)
Elephant foot compensation: 0.3 (from 0.1)

suchmememanyskill, 2023-11-12:
I personally use adaptive cubic infill all the time already, so i'm ok with that one

3 perimiters i'm a little unsure on, as it's only really required for strong parts. It'd waste filament for people who only print decorative models

Elephants foot i feel like changes from printer to printer, i get very good results at .1

suchmememanyskill, 2023-11-14:
Closing due to cubic infill being the default now

closed 2023-11-14 by suchmememanyskill