suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config #12: Please add Extruder Clearance for sequanital printing

foobnix, 2023-11-15:
Please add correct Extruder Clearance for sequanital printing

https://help.prusa3d.com/article/sequential-printing_124589

I am using 45x45 but it's not optimal

suchmememanyskill, 2023-11-15:
I wonder how useful it is to set the value correctly in general, see the bottom of that page. You'd need to ajust the values yourself anyway if you print tall objects but put them along the Y axis (where after the print is done, the y bed moves backwards and thus the X axis will never hit the print)

suchmememanyskill, 2023-11-17:
The extruder shroud's longest side is roughly 75mm. This makes a safe radius around 53mm. The available height is roughly 40mm.

Entering these values will put large gaps around your parts. Space that can be better used if parts are just positioned in a smart way (behind the print head). I don't think this is a useful addition, as it'll only get in the way of being able to position elements sequentially behind the print head

suchmememanyskill, 2024-01-07:
Closed due to inactivity

closed 2024-01-07 by suchmememanyskill