suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config #14: Klipper user bed 230x225

NotEaZy, 2023-12-09:
Hi! I stumble accross you prusa slicer profile and it have been a great help to me getting the speed and print quality. 

However on your release you stated that "Klipper users should double check if their config allows for a 230x225 bed".
I believe i will have dimensional issue if the problem not arrest.

Recently i just migrated from marlin to klipper and still learning. 

Do i change my config stepper_x max to 230 or 244 (my max x postion)? If it is possible can i ask for your printer.cfg enable me to compare with mine.

Thank you!

suchmememanyskill, 2023-12-09:
You won't have dimensional issues if you'd run that bed size while you don't support it. Rather, if you print too far out, klipper will stop you.

My config is based on https://github.com/0xD34D/ender3-v3-se-klipper-config/blob/main/printer-creality-ender3-v3-se-2023.cfg#L30 . See the stepper x, stepper y and bltouch sections. All need to be edited to get a 230x230 bed (for y, 225 is used for some safety margin on the screws in the back).

Or, you can just set the bed size back to 220x220. Up to you

NotEaZy, 2023-12-09:
Thank you for your clarification. Please keep up the excellent work. Thanks!

closed 2023-12-09 by NotEaZy