suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config #19: Print info preview

hamzag250, 2023-12-16:
Is it possible to get print info like print time, filament usage and other
It actually work in cura but cant find any tweak to get it work in prusa, otherwise your work is fantastic and print quality is exellent

suchmememanyskill, 2023-12-16:
I have no idea. I don't run official firmware anymore. I suspect it's done using gcode comments at the start of the file but prusaslicer inserts preview images there

hamzag250, 2023-12-16:
It's been a gcode comment that is on the first lines:
;FLAVOR:Marlin
;TIME:XXXXX (in second)
;Filament used: XX.XXXXm
;Layer height: 0.2
can't find any output option to get this on the very first line of gcode in prusa

ottelo9, 2023-12-31:
@suchmememanyskill 
Is there a mod firmware for the printer?

@hamzag250 
Maybe with a script?

suchmememanyskill, 2023-12-31:
With a script it would be possible to replicate, although that sounds very janky

suchmememanyskill, 2024-01-07:
@hamzag250 @ottelo9 so after a bit of trial and error, i made a post processing script to add pre-print info to the screen

```
import sys

filename = sys.argv[1]

with open(filename, 'r') as fp:
    data = fp.readlines()

filament_used_mm = 0
time_s = 0
layer_height = 0

for line in data:
    if line.startswith("; filament used [mm] = "):
        filament_used_mm = float(line[23:])
    if line.startswith("; estimated printing time (normal mode) = "):
        line_times = [x.strip() for x in line[42:].split(" ")]
        for time in line_times:
            if time.endswith("h"):
                time_s += int(time[:-1]) * 3600
            elif time.endswith("m"):
                time_s += int(time[:-1]) * 60
            elif time.endswith("s"):
                time_s += int(time[:-1])

    if line.startswith("; layer_height = "):
        layer_height = float(line[17:])

print("Filament used: " + str(filament_used_mm) + "mm")
print("Time: " + str(time_s) + "s")
print("Layer height: " + str(layer_height) + "mm")

extra_data = [
    ";FLAVOR:Marlin",
    f";TIME:{time_s:.2f}",
    f";Filament used:{filament_used_mm/1000:.5f}m",
    f";Layer height:{layer_height}",
    ";",
    ";;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;"
]

with open(filename, 'w') as fp:
    fp.writelines([x + "\n" for x in extra_data])
    fp.writelines(data)
```

Enjoy!
I'll probably write some docs on how to actually make use of this

hamzag250, 2024-01-07:
It work fine for me. It might be useful to open his own repository with tutorial and quote it on this one

suchmememanyskill, 2024-01-07:
> It work fine for me. It might be useful to open his own repository with tutorial and quote it on this one

Already added an extra .md file to this repo with instructions, but haven't referenced it yet

suchmememanyskill, 2024-01-07:
Closing as released

closed 2024-01-07 by suchmememanyskill