suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config #23: layer 2 speed ignored when sequentially printing objects

BastiSch, 2023-12-30:
When sequentially printing multiple objects, the layer 2 (and layer 3) speed settings are ignored.
I believe this is not intended, though I am unsure what the main reason behind the lower speed for layers 2 and 3 is and have not gathered enough experience to judge whether it is necessary.
Anyway, I have changed it in my configuration and wanted to share this.

Accroding to my observations `layer_num` is not reset between objects but simply further incremented for the second (and further) objects continuously. So it cannot be used to derive whether this is the second layer reliably. This is why I changed the G-code to derive the speed using `layer_z` instead.

I use the following G-code, which sets the speed for each layer in "After layer change G-code":
```
{if layer_z <= first_layer_height}M220 S100 ; first layer speed is already handled in the presets
{else}M220 S{round( min(100, 25 + 25*(layer_z - first_layer_height)/layer_height ) )}; set speed above the first layer increasing from 50% in layer 2 to 100% in layer 4 and higher
{endif}
```
This additionally adds the M220 before each layer explicitly (also layer 1 and all layers > 4). If this is unwanted, one could also just replace `layer_num` by `round(1 + (layer_z - first_layer_height)/layer_height)` to have exactly the same as before for single object (or non-sequential) prints.

suchmememanyskill, 2023-12-31:
I added the 2nd/3rd layer slowdowns as other slicers like cura and creality print have this too. It supposedly isn't a good idea to print fast on a possibly unstable and uneven first layer. I haven't needed this but it doesn't hurt either

I also have observed this issue. My idea was to look through the docs to see if there's some per item z height that could be used

I would prefer to not set it every layer, just to avoid wrong input (like what if the formula freaks out at line 300 specifically?). Nice find, though

BastiSch, 2023-12-31:
I agree to the small risk of setting it on every layer.
However, not setting it on every layer adds another small risk that if a layer at the height of layer 4 is missing (I would not imagine how though) it would continue at 75% (which would still be noticeable as the predicted printing time takes it into account).

To reduce the risk one could also use it only below a certain height. Maybe like this:
``` 
{if layer_z > first_layer_height && layer_z <= first_layer_height + 3.5*layer_height}M220 S{round( min(100, 25 + 25*(layer_z - first_layer_height)/layer_height ) )}; set speed above the first layer increasing from 50% in layer 2 to 100% in layer 4 and higher
{endif}
```
This would only insert it for layers at the usual height of layers 2 to 4. (I also noticed with thin supports there might be layers in between the usual heights.)
Another possibility would be staying closer to the original version:
```
{if round( (layer_z - first_layer_height)/layer_height) == 1}M220 S50 ; set speed to 50% for layer 2{endif}
{if round( (layer_z - first_layer_height)/layer_height) == 2}M220 S75 ; set speed to 75% for layer 3{endif}
{if round( (layer_z - first_layer_height)/layer_height) == 3}M220 S100 ; set speed to 100% for layer 4{endif}
```
Or getting rid of using `round` again by:
```
{if layer_z < first_layer_height + 0.5*layer_height}; first layer is already handled by preset
{elsif layer_z < first_layer_height + 1.5*layer_height}M220 S50  ; Set speed to 50% for layer 2
{elsif layer_z < first_layer_height + 2.5*layer_height}M220 S75  ; Set speed to 75% for layer 3
{elsif layer_z < first_layer_height + 3.5*layer_height}M220 S100 ; Set speed to 100% for layer 4
{endif}
```
Remark: I am not confident on using `layer_z == first_layer_height + layer_height`, as this is usually a bad thing to do with floats.

Just some ideas. Would be simpler if prusaslicer would have a setting like "Number of Slower Layers", as Cura has.

suchmememanyskill, 2024-01-07:
I think your last block is actually great. Do you mind if include it and credit you in the release notes for it?

suchmememanyskill, 2024-01-07:
Proposed gcode:
```
{if complete_objects == false}

{if layer_num == 1}M220 S50  ; Set speed to 50% for layer 2
{elsif layer_num == 2}M220 S75  ; Set speed to 75% for layer 3
{elsif layer_num == 3}M220 S100 ; Set speed to 100% for layer 4
{endif}

{else}

{if layer_z < first_layer_height + 0.5*layer_height}; first layer is already handled by preset
{elsif layer_z < first_layer_height + 1.5*layer_height}M220 S50  ; Set speed to 50% for layer 2
{elsif layer_z < first_layer_height + 2.5*layer_height}M220 S75  ; Set speed to 75% for layer 3
{elsif layer_z < first_layer_height + 3.5*layer_height}M220 S100 ; Set speed to 100% for layer 4
{endif}

{endif}
```

I sadly don't see an option to check what layer type you're working with. Nor an option to fix variable layer height shenanigans. Although the speed being set more than once isn't the end of the world, i suppose

BastiSch, 2024-01-07:
> I think your last block is actually great. Do you mind if include it and credit you in the release notes for it?

Feel free to use it. This what I shared it for.

> Although one last thing to fix to be done: This doesn't work when variable layer height is used or when supports are used

Getting it to work perfectly for variable layer heights and support with all different options will probably not be possible (with justifiable complexity of the code). But even then it will print some layers slower improving the print.

suchmememanyskill, 2024-01-07:
Closing as released

commit caa42278 (2024-01-07): Fix #23
closed 2024-01-07 by suchmememanyskill