suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config #29: No prepping

andersborgabiro, 2024-01-18:
When I used the Ender-3 profile provided with PrusaSlicer it painted a line in the bottom-left corner (once down, once up), so that the extruder is prepped before the actual printing.

This profile bundle doesn't do that. Instead it just sets the extruder at the height of the bed at the bottom-left corner while heating up to the target temperature, which (I'm guessing) could potentially damage the bed, and also there's no prepping.

I noticed that the original Ender-3 profile had 0.08 as the finest layer height, while this one has 0.10. Is 0.08 indicated as too low for this printer?

suchmememanyskill, 2024-01-18:
If your printer is calibrated well, then it shouldn't damage the bed. If it does, it likely would have done so too during the actual print

There is prep, in the form of a skirt around the print. The length of the skirt has been doubled i believe from the original ender 3 profile.

I chose 0.1 as the minimum line height as i don't think most want to realistically go any lower. But the dropdown layer height options are only there for convenience. If you want to change it, change 0.1 to 0.08 in the print settings

suchmememanyskill, 2024-01-18:
Creality print's start gcode actually does the same, parking the print head in the corner very close to the bed. I did the same as i noticed filament would droop out and get stuck to the nozzle before; right now it gets stuck to the bed.

The nozzle gets parked 0.28mm above the print bed, btw

andersborgabiro, 2024-01-18:
OK. It's fine then.

closed 2024-01-18 by suchmememanyskill