suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config #31: Python script not running properly

Bentheminernz, 2024-02-02:
When prusa goes to run the gcode editor python script, it fails on my mac showing this error.
`Post-processing script /Users/benlawrence/Documents/3dPrinting/PrusaSlicerEnder3v3SETimeScript/info.py on file /var/folders/42/wztjy5ws2f79cqk2fhcv64400000gn/T/.21217.gcode.pp failed.
Error code: 2
Output:
/Users/benlawrence/Documents/3dPrinting/PrusaSlicerEnder3v3SETimeScript/info.py: line 1: import: command not found
/Users/benlawrence/Documents/3dPrinting/PrusaSlicerEnder3v3SETimeScript/info.py: line 2: import: command not found
/Users/benlawrence/Documents/3dPrinting/PrusaSlicerEnder3v3SETimeScript/info.py: line 4: filename: command not found
/Users/benlawrence/Documents/3dPrinting/PrusaSlicerEnder3v3SETimeScript/info.py: line 6: syntax error near unexpected token `('
/Users/benlawrence/Documents/3dPrinting/PrusaSlicerEnder3v3SETimeScript/info.py: line 6: `with open(filename, 'r') as fp:'`

suchmememanyskill, 2024-02-02:
You seemingly forgot to add path/to/python in front of the script path.

Maybe it is a useful addition to add a shebang though

suchmememanyskill, 2024-02-18:
Closing due to inactivity

closed 2024-02-18 by suchmememanyskill