suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config #32: blobs and zits when printing

keysCodes, 2024-02-02:
been using this profile perfectly fine for a while then this started happening
[image] theres also a lot of zits around a lot of different prints

keysCodes, 2024-02-02:
also i got the nebula pad and when i got it the prints started out perfect but after a couple weeks it stopped coming out perfect so its prob not a pad issue (i use the klipper version on the cfg because nebula has klipper)

suchmememanyskill, 2024-02-02:
By that same logic, why would this profile be the issue then, if it worked before?

It's possible your filament isn't dry, please do check that

keysCodes, 2024-02-06:
yes i have i have dehydrated it many times and its not brittle so

keysCodes, 2024-02-06:
also it works with other slicers just that they have other problems but the prusa profile before it had issues didnt

suchmememanyskill, 2024-02-06:
What does 'before' and 'after' mean in this context? If the profile didn't change i'm unsure how this is my issue

keysCodes, 2024-02-07:
before it stopped working and after

suchmememanyskill, 2024-02-07:
> If the profile didn't change i'm unsure how this is my issue

Closing this issue, this seems to be an issue dictacted by outside factors, not the profile

closed 2024-02-07 by suchmememanyskill

keysCodes, 2024-02-11:
do you atleast know of a possible fix?