suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config #35: M420 S1  "restore bed leveling" command is needed after G28 home command in Start G-code

kardashov, 2024-03-31:
Hi,

According to Marlin docs https://marlinfw.org/docs/gcode/M420.html

> "By default G28 disables bed leveling. Follow with M420 S to turn leveling on. "

Which is the case for this printer. Without M420 S1 command printer doesn't take bed leveling data into account and may produce problematic first layer (unless you have ideally leveled bed).

I suggest to add this command to this profile, this greatly improved first layer, adhesion and overall quality of my prints:

https://github.com/suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config/blob/main/PrusaSlicer/printer/Ender%203%20V3%20SE.ini#L71

[image]

suchmememanyskill, 2024-03-31:
> Which is the case for this printer. 
How do you know this is the case for this printer?

Also, would you mind sending out a pull request for this?
Also, wouldn't it be better to just do a G29 (level) then?

kardashov, 2024-04-01:
I've discovered issue when tried to troubleshoot adhesion problems for parts in certain areas of my print bed and stumbled upon discussion [on Reddit](https://www.reddit.com/r/klippers/comments/pco17y/do_we_need_to_use_g29_or_m420_s1/). It is related to Klipper but also applies to Marlin.

Print bed level deviations of +/- 0.3mm are considered normal (green on UI) by Ender's auto leveling software and can be "fixed" by printer on the fly. But even Creality Print's Ender 3 V3 SE profile has the same problem and produces gcode which does not take leveling data into consideration after G28 Home resets it.  It is likely they forgot to enable this parameter during compilation:
> By default G28 disables bed leveling. Follow with M420 S to turn leveling on.
    With ENABLE_LEVELING_AFTER_G28 leveling will always be enabled after G28.
    With RESTORE_LEVELING_AFTER_G28 leveling is restored to whatever state it was in before G28 

This leads to not perfect first layer (barely visible adhesion problems or under extrusion in certain areas) and even ruined prints for details with small footprints.

> Also, wouldn't it be better to just do a G29 (level) then?

G29 is for creating a new mesh while M420 S1 is for loading an existing one.

Here is PR https://github.com/suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config/pull/37

suchmememanyskill, 2024-04-01:
> G29 is for creating a new mesh while M420 S1 is for loading an existing one.

Yeah i'm aware what it does, but why pick one over the other?

Imo, for ease of use, it may be a better idea to always do a mesh bed level before a print. Just in case there's any user error in setup.

Also thank you for the PR!

mpaterakis, 2024-04-30:
This fixed first layer (and therefore subsequent) issues for me, glad to see the [PR](https://github.com/suchmememanyskill/PrusaSlicer-Ender3-v3-SE-Config/pull/37) merged. It should be added to a release asap.

I kept auto-leveling my bed without any real difference, and always had to manually set my z-offset because the auto-level's was completely "off"; my offset was fine as it turns out, and it levelling was never applied. This is a major feature of the V3 SE, I'm shocked that it wasn't used from the get-go.

As for re-levelling on every print, that's **way** too excessive. If at any point you feel that your level is not as it should, you can just manually re-run it through CR-Touch. The PR change is more than enough.

suchmememanyskill, 2024-04-30:
Released

closed 2024-04-30 by suchmememanyskill